PaxInstruments/t400-enclosure #60: High-Cycle, Longer-Life Bail Design!!!

starno, 2015-01-02:
Here's an idea for an alternative approach at the bail/battery cover pair.  I think this is more promising for repeated cycles, and maintains the lock-open and lock-closed requirements.  It also can be made with a single-pull mold.

It works by having a flexible "wing" on the bail with a boss on the end:
[image]
There are corresponding holes on the battery cover to control it's position.

In the closed position:
[image]

As it opens, the wing flexes against the wall and creates resistance:
[image]

In the open position a notch on the battery cover wall allows the wing to return to its natural position, and prevents the bail from closing without user pushing on it:
[image]

If the bail is opened beyond its natural open position, it stops against the wall like the current bail:
[image]

Thoughts?

charlespax, 2015-01-04:
Looks good. With the right hold placement the bail should feel really solid.

How do you feel about getting some parts machined to try this out? We would need the battery cover, bail, and part of the enclosure back. We just need enough to test this with actual ABS. Better to test the mechanism early than to spend time fully implementing it only to find we don't like it.

starno, 2015-01-22:
I feel good about it.  CNCd out of ABS should be really telling for how well it will work.  This has been merged into the original files and replaced the old bail in all the CAD master files.

charlespax, 2015-01-22:
Do we need any chamfer on the highlighted edges?
[image]

starno, 2015-01-24:
The bail and battery cover assembly had a lot of refinements in 49284562307c798eb90a043f63ebbd4d5f7dcef9

Those particular edges have a drafted surface that acts like a chamfer.
[image]

closed 2015-01-24 by starno