PaxInstruments/t400-enclosure #61: M2 or M3 screws?

charlespax, 2015-01-03:
In my original prototype I used M2 screws, which has carried over to the production design. I think it might be good to make explicit decision about this. @starno do you think the M2 screws be strong enough or would we need M3 screws?

starno, 2015-01-03:
I like the M2s, they dont take up a lot of space, and the inserts are tiny.  Do any of the prototypes feel insecure?

charlespax, 2015-01-03:
The threaded inserts didn't melt into the Enclosure version 0.1 (SLA) print because of the geometry (https://github.com/PaxInstruments/t400-enclosure/issues/39) and I think the material. The PLA prototypes I made months ago always felt fine.

starno, 2015-01-03:
I see.  For the insert bosses that do not fit, try the threaded inserts I bought, they have a smaller diameter and seem to be more abundant.  The mismatch with the battery cover has to do with opposite draft direction.  In fact, the majority of issues remaining are a result of drafting changing clearances, and I need to consult some friends on the best approach for this situation.  I think what needs to be done is breaking out some math and change the dimensions based on length of boss and angle of draft.

starno, 2015-01-22:
Fasteners added
[image]

closed 2015-01-22 by starno