PaxInstruments/t400-enclosure #62: MicroUSB port squishes keypad

charlespax, 2015-01-03:
I put the Electronics version 0.9 PCBA in the Enclosure version 0.1 (SLA) print. In the image below you can see how the MicroUSB connector pressed directly against the keypad. Since the keypad in this print is hard SLA material, the buttons do not work. This may not be an issue in the silicone keypad.

[image]

starno, 2015-01-22:
As of now the buttons and connector are the same height, so it might affect the feel of 2/3 buttons on that side.  One option is to get taller buttons and a make a corresponding indent in the silicone to accommodate.  @charlespax  @samchoy88 Do we have a final button configuration/part number chosen already?

charlespax, 2015-01-22:
Here is a switch with a taller button. The height from PCB is 3.63 mm unpressed and 3.43 mm when pressed.

[image]
[image]
[image]

starno, 2015-01-22:
Ok, thanks.  That's going to interfere with the silicone buttons so I will add a recess
[image]

charlespax, 2015-01-22:
Will the distance between the keypad bottom and the microUSB connector increase? If not, I think we'll still have the squishing issue.

[image]

starno, 2015-01-22:
It did not increase but is able to.  I don't think the squishing issue will still be there because the silicone rests directly on the buttons and will be depressed before there is any contact with the usb connector.
[image]

charlespax, 2015-01-23:
I think the USB connector model may not be accurate. The gap that exists in the assembly did not appear to exist in the prototype. Can you physically verify there is a gap I. The prototype you have?

starno, 2015-01-23:
Good call, there was some interference there

there should be plenty of clearance there now
[image]

closed 2015-01-23 by starno