PaxInstruments/t400-enclosure #64: Fastening PCB

starno, 2015-01-17:
Based on the discussion @charlespax we decided to put the pcb onto bosses with a shoulder, the boss  passes through the PCB and the corresponding boss on from the back will meet up with it and keep everything constrained.

unfortunately two problems arise:
[image]

The front panel bosses are larger than the back because they need to contain the threaded inserts.  This makes the hole in the PCB larger and the back panel boss diameter is not large enough to prevent any unwanted movement of the PCB.  We either have to reconsider the assembly operation or use the pinching method like before.

Thoughts?

charlespax, 2015-01-17:
I think shoulder bosses on the back is better than no shoulder. I don't think it would affect the assembly process since the LCD is already putting the PCB in the right place.

starno, 2015-01-17:
I agree.  I also did a test run of the assembly with the prototype models and if the fit is tight on the PCB putting the back on with the PCB on it doesn't make a huge difference.  Either way I'll switch up the configuration

charlespax, 2015-01-17:
Cool.

starno, 2015-01-22:
This was fixed in commit bc3faf97fd03e1f0dbca54480562c21bee565d81
[image]

closed 2015-01-22 by starno

charlespax, 2015-01-22:
The PCB mounting holes need to be changed, issue opened (https://github.com/PaxInstruments/t400-electronics/issues/139).

@starno please comment on the issue (https://github.com/PaxInstruments/t400-electronics/issues/139) with the new hole dimensions.