PaxInstruments/t400-enclosure #65: Tilting bail lateral movement

charlespax, 2015-01-22:
It looks like the bail will be able to move left and right quite a bit. 

[image]

Should the two surfaces marked in yellow mate?
[image]

charlespax, 2015-01-23:
@starno what do you think?

starno, 2015-01-23:
No, I don't think they should mate.  The bail and battery cover still need finishing touches.  will report back

starno, 2015-01-24:
Ok, have a look at the files from 49284562307c798eb90a043f63ebbd4d5f7dcef9

Lateral movement is protected by the non-nub sides of the hinge in the center of the bail.  The length of both the axle boss and the locking wing boss both increased for strength and preventing accidental removal.
[image]

charlespax, 2015-01-25:
Cool. Looks like 0.26 mm. Let's close this out and readdress if the Enclosure version 0.2 (SLA) prototype has any problems.
[image]

closed 2015-01-25 by charlespax