PaxInstruments/t400-enclosure #66: Tilting bail flexing

charlespax, 2015-01-22:
When the tilting bail is snapped into the battery cover the tilting bail must flex. Are the cutouts long enough to allow this?

[image]

starno, 2015-01-22:
so the thought here is that the cutout length needs to be on the shorter side of acceptable for two reasons:  it's a one-time snap; it must be harder to flex than the "wing" to avoid the bail from displacing or falling out when trying to open or close the bail.  The cnc parts will be telling.  Do you want to start longer than it is?

charlespax, 2015-01-22:
For the cutout length we should just machine what you have here. We can always hacksaw it longer to find the perfect length. However, the cutout width may not be enough. The cutout width is 1 mm. The nub is engaging with the battery cover 1 mm. It would be bad if the flexing part bottoms out before the nub disengages from the battery cover.

charlespax, 2015-01-23:
@starno can you do some measurements or FEA to see if what you have will work?

starno, 2015-01-24:
According to FEA calcs, a 40N (~9+lbf) load on the hinge nub will displace it the necessary 1mm to insert into the battery cover.  
[image]

At that point the largest stress point is indicated in below in red, and is only half the yield strength of ABS.  So it looks pretty close to what we would want.
[image]

closed 2015-01-24 by starno