PaxInstruments/t400-enclosure #67: top panel orientation key

charlespax, 2015-01-26:
Right now the top panel is capable of being inserted in any of four orientations. We can add a small key onto the top panel and a keyway into the enclosure to ensure it can be inserted in only one orientation. This would not compromise the ability to use a 1.6 mm PCB as a top panel.

The notch would go on the front side or back side based on assembly order.

It could look something like this.
[image]

starno, 2015-02-02:
good call.  added two small locators on edge of top 

[image]

charlespax, 2015-02-10:
The top panel can be inserted the wrong way. I'd have to check the SLA print, but I believe the enclosure cannot be screwed shut if it is wrong. Not a huge problem, but a worker would have to correct the mistake. We can avoid them making a mistake entirely if we add a single keyway to the center on the other side.
[image]
Reopening issue and moving to Enclosure version 0.4 (SLA).

starno, 2015-03-20:
fixed. 

[image]

closed 2015-03-20 by starno