PaxInstruments/t400-enclosure #71: Gap around enclosure

charlespax, 2015-01-30:
There is very little gap around the enclosure. This is the gap around the outside separating the front and back.

starno, 2015-02-01:
@charlespax This really hasn't changed from previous prototypes.  The gap there is 0.25mm, what would you like it to be?

[image]

charlespax, 2015-02-01:
To make a decision on this I will look at a few devices to get a sense of what looks good. I'll post my response here.

charlespax, 2015-02-10:
Woops! I didn't get to this in time. Moving to Enclosure version 0.4 (SLA).

charlespax, 2015-02-19:
Let's do 0.25 mm in the next SLA print and see how it looks.

starno, 2015-03-09:
@charlespax that means it will go unchanged.  should I close this issue?

closed 2015-03-09 by charlespax