PaxInstruments/t400-enclosure #72: Top panel gap

charlespax, 2015-01-30:
Should we have a gap just like the outer edge (https://github.com/PaxInstruments/t400-enclosure/issues/71)?

starno, 2015-02-01:
I think both gaps should match

charlespax, 2015-02-01:
I agree. In the next version whatever we pick for the gap around the enclosure (https://github.com/PaxInstruments/t400-enclosure/issues/71#issuecomment-72352186) we should have for the panel.

charlespax, 2015-02-10:
Woops! I didn't get to this in time. Moving to Enclosure version 0.4 (SLA).

charlespax, 2015-02-19:
Looking at a few devices I thin 0.25 mm will be good. Let's to that and see what we get in the next SLA print.

charlespax, 2015-03-19:
Looks fine to me. Can we close this?

[image]

starno, 2015-03-20:
yep.  its at 0.25mm as desired

closed 2015-03-20 by starno