PaxInstruments/t400-enclosure #79: SD card slot intererence

charlespax, 2015-01-30:
In Enclosure version 0.2 (SLA) the SD card slot does not match up with the enclosure. The card slot also interferes with the enclosure.

starno, 2015-02-01:
There is definitely some discrepancy here between the CAD and the prototype board.  The connector on the prototype measures a 1.3mm overhang from the pcb edge, when it should be 0.7mm.  (discussed in issue #53)
[image]
@samchoy88 @charlespax can you verify the overhang of the SD connector on the PCB is 0.7mm and the distance from the edge of the pcb to the centerline of the SD connector is 23.51mm?  
[image]
If it is easier for me to shift the opening on the enclosure, I can do that too.  Let me know

charlespax, 2015-02-01:
In eagle I get 1.45 mm overhang. Opened issue  https://github.com/PaxInstruments/t400-electronics/issues/144

[image]

starno, 2015-02-01:
Please check the the distance from the side of the PCB as well since the alignment with the opening looks to be around 1mm off

charlespax, 2015-02-01:
Here you go.

[image]

starno, 2015-02-01:
Ok, that looks correct.  I must have the SD placement within the connector modeled wrong

starno, 2015-02-01:
Yup, that was it, and explains the shifted hole in the enclosure.  I had the SD card 0.2mm from the left edge of the connector when looking at its opening.  In the physical connector, this measures to be 1mm.  This was the missing information I mentioned https://github.com/PaxInstruments/t400-electronics/issues/79#issuecomment-68165430   The opening on the enclosure is based on the location of SD card and not the connector location, so it explains the shift.  If https://github.com/PaxInstruments/t400-electronics/issues/145 exposes some other dimensions that are off between the library component and the one we are working with, then we can reopen this issue.

closed 2015-02-01 by starno