PaxInstruments/t400-enclosure #80: PCB locating nubs

charlespax, 2015-01-30:
We can have two little nubs to help locate the PCB during assembly.

starno, 2015-02-01:
Locating features added.  They ride up the sidewalls of the front of the enclosure and pass through the PCB on either side.  This constrains the PCB during assembly in the Y direction, and there is a third rib extension that prevents the pcbs from going too far left.  The right side is prevented because the USB connector is in the way.
[image]
There is less than 0.1mm between the PCB and locating ribs after the draft.

An important thing to note is that the silicone buttons are the same part on both sides.  There is a chunk cut out of them that allows the locating ribs to pass through and this doubles as a way to prevent the silicone pieces from going in upside down.  However, during the icon printing process, the left and right sides need to be printed in opposite orientations.
[image]

charlespax, 2015-02-01:
We can't have a rib against the edge of the PCB because that edge will likely be a v-groove and have poor dimensional tolerance. The milled slots work well because we have good control over those dimensions.

charlespax, 2015-02-01:
Also, we should clear some space around the microUSB connector. We discussed a chamfer to give some clearance.

starno, 2015-02-01:
Good call.  I removed that locating rib against the pcb edge, reverted the larger USB opening to what it was and added a chamfer on the inside edge instead.

[image]

closed 2015-02-01 by starno