PaxInstruments/t400-enclosure #82: Injection mold STLs for 3d printing

cibomahto, 2015-02-22:
It would be awesome if there were STL files available for 3d printing a case based on the injection mold design. The solidworks parts are awesome but pretty huge to download over slow internet.

charlespax, 2015-02-22:
There are older set of  Solidworks and STL files for 3D printing in https://github.com/PaxInstruments/t400-enclosure/tree/master/3D%20Printed. After we go into production, we should update or redesign 3D printable files.

charlespax, 2015-02-22:
@starno has been having pretty good success with FDM 3D prints. The only significant problems is the bridging involved with the battery compartment.

cibomahto, 2015-02-22:
Yeah, no rush here- sometime after everything is finalized is probably soon enough :-).

charlespax, 2015-02-26:
We'l get to this at some point. Putting in ice box.

starno, 2015-02-26:
I've kept STLs out of the repo during development because the design is changing a lot and they quickly become useless, making the repo even larger.  As of now the only thing that has trouble with FDM are the small features on the bail.  Those features are going to be getting larger and if the production units are not easily printed, I can make a printable version.

charlespax, 2015-03-19:
At this point I don't think we need to commit to doing anything special. Let's see how the final design works for printing. Moving to ice box.

charlespax, 2015-08-08:
3D printing technology is good enough that people can order SLA prints of the actual enclosure. If we make another one that is good for 3D printing, I would consider it a separate product. Closing issue.

closed 2015-08-08 by charlespax