PaxInstruments/t400-enclosure #83: Logo style

charlespax, 2015-03-01:
I think the logo on the enclosure is a bit weak with just the P-microscope. I'd like to try the logo with the circle and with the rounded square as seen in the images below. The outer shape would be an extrude cut into the enclosure. The P-microsope would be an extrude from the bottom of the extrude cut to the surface of the enclosure.

[image]

charlespax, 2015-03-01:
I've added Adobe Illustrator files to the repo.

starno, 2015-03-09:
First impressions is the rounded square will match the shape of what we got already.  I will post a comparrison

starno, 2015-03-09:
Hmm, here is a quick comparrison of the two:
[image]
[image]
I'm not so sure I like the border there because the logo needs to get much smaller to keep the same margin proportions.  My personal preference would be to emboss the logo there and maybe add some color to make it pop.  @charlespax what do you think?

starno, 2015-03-09:
Something kinda like this: 
[image]

charlespax, 2015-03-09:
I agree with you. I guess we're down to two choices: colored logo on the enclosure or logo moved from enclosure to the front label. Do you have a string opinion either way?

charlespax, 2015-03-09:
If the logo were in the label, it would be smaller than a label embossed on the enclosure.

MLXXXp, 2015-03-09:
I don't know if you even care, but if the logo is on a stick-on label it makes it easier to OEM the product with a different brand. Embossing the enclosure makes it's more difficult to change. It's also easier if you decide to change your own company logo in the future.

You could also consider just adding a separate larger indent specifically for a logo sticker.

charlespax, 2015-03-18:
@MLXXXp I agree. We've discussed this before and I think that's a fairly compelling reason. Combined with my desire for flexibility, we should kill the embossed logo and add it to the front sticker.

@starno Let's kill the enclosure logo and move it to the sticker.

charlespax, 2015-03-19:
I just did a git pull and opened the file. The logo is still there for me. Am I missing something?
[image]

charlespax, 2015-03-19:
My fault. I was looking `master` rather than `starno`.
[image]

Looks done. Closing issue.

closed 2015-03-19 by charlespax