PaxInstruments/t400-enclosure #84: Add rib to snap wings

starno, 2015-03-09:
The snaps are a bit weak at the base and showing some stress marks.  a small rib reinforcement should help here

[image]

charlespax, 2015-03-19:
With the other strength changes do you think you need to do this?

starno, 2015-03-20:
I think we can hold off on it.  Like you mentioned the snaps on the bail got a lot longer and stronger, so we can hold off unless it's needed again.

[image]

charlespax, 2015-03-20:
Okay. Moving this to ice box. If it's not a problem with the next prototype, we should close it.

charlespax, 2015-08-08:
Moving to 0.6 as something we should test when we get the prototypes back.

charlespax, 2015-08-10:
Added to the review checklist. Closing issue. We can reopen if necessary after test.

closed 2015-08-10 by charlespax