PaxInstruments/t400-enclosure #85: Shallower angle on the snap wings

starno, 2015-03-09:
extending the snaps to span further along the edge of the bail can both strengthen the bail and wings

[image]

charlespax, 2015-03-18:
Since you're having trouble with batter interference with that design maybe this a viable option:
[image]

I don't think this interferes with anything.

charlespax, 2015-03-19:
Can we close this?

starno, 2015-03-20:
yeah, bail got a lot stronger

[image]

closed 2015-03-20 by starno