PaxInstruments/t400-enclosure #86: Further tighten bail clearance inside cover

starno, 2015-03-09:
There is still too much wiggle room allowing it to slide back and forth when jostled

charlespax, 2015-03-19:
Looks good in the model. Can we close this?

starno, 2015-03-20:
@charlespax Im tempted to add in some bulbous features on the inside edge of the bail to give it a point contact with the battery cover and have minimal friction when opening.  The clearance between the two parts is tighter than the last prototype, but I'm starting to think something along these lines is the only way to eliminate it.  What are your thoughts?

[image]

charlespax, 2015-03-20:
That makes sense. Maybe put them on the most flexible parts of the tilting bail.

[image]

starno, 2015-03-20:
moving this to the production unit milestone.  Depending on how 0.4 comes out, this is something to iron out with tooling and make sure things are cut short so we can always add a bit of material until the fit is right.

charlespax, 2015-03-20:
Moved to milestone 0.5. The tooling milestone is only for things relevant to the tooling.

charlespax, 2015-08-10:
Okay, let's see if this problem comes up again in the next prototypes. If so, we can file a new issue. Closing issue.

closed 2015-08-10 by charlespax