PaxInstruments/t400-enclosure #87: Keep snaps from escaping battery cover's grasp

starno, 2015-03-09:
Currently the bail "locks" open when the snaps pop out of the battery cover and sit in an open pocket.  
[image]
The prototype is difficult to "unlock" from this position when closing the bail, however it is very easy to open back up again.  Therefore, the bail hinge needs to sit a little deeper inside the battery cover to prevent the snaps from ever becoming totally free.

charlespax, 2015-03-19:
Looks about right in the model. Can we close this?

starno, 2015-03-20:
Yeah, bail snaps managed to get longer, the locking positions moved toward the top, and the open locking position is recessed to prevent over extension.

[image]

closed 2015-03-20 by starno