PaxInstruments/t400-enclosure #88: Top panel holes

charlespax, 2015-03-19:
We are developing a custom thermocouple connector (https://github.com/PaxInstruments/t400-electronics/issues/17). We may have to move the connector holes on the top panel to match. There is a 3D model in the electronics repository (https://github.com/PaxInstruments/t400-electronics/tree/master/solidworks/clip).

Current design:
[image]

starno, 2015-03-20:
I like this because it moves the holes in the top closer to true center.  I updated the PCBA to have these populated, but I left them in the same locations as the previous connectors.  

[image]
resulting top panel:
[image]

@charlespax do you want smaller number indicators now?
@samchoy88 do you have coordinates of each thermocouple connector for me so I can match the PCBA in CAD to match the latest board files?

charlespax, 2015-03-20:
I don't have the coordinates for the connectors. Once the connector design is approved and we cut a PO, we will have an accurate model. The connector is still in the design phase.

charlespax, 2015-03-20:
The number can probably be a touch smaller, but I think it looks fine this way too. Let's get a prototype of this design and see how we feel about it.

charlespax, 2015-04-02:
The most recent prototype looked great. Closing issue.

closed 2015-04-02 by charlespax