PaxInstruments/t400-enclosure #92: Silicone keypad graphics

charlespax, 2015-03-23:
@paxmanchris is taking care of the silicone keypad graphics.

The graphics can be found in the manual https://github.com/PaxInstruments/t400-electronics/blob/master/documentation/manual.md

Here are the dimensions we are working with.
[image]

paxmanchris, 2015-03-23:
Here is a draft for the cycle icon
[image]

charlespax, 2015-03-23:
[image]

Cycle channels
[image]

Backlight 
[image]

Power
[image]

Logging start/stop
[image]

Log interval
[image]

Units: Unknown

paxmanchris, 2015-03-23:
Okay, I [uploaded the design](https://github.com/PaxInstruments/t400-enclosure/blob/master/graphics/buttons_worksheet.svg), staying as close to what was described in the [documentation](https://github.com/PaxInstruments/t400-electronics/blob/master/documentation/manual.md#buttons). Here is a picture what I have done so far:

[image]

@charlespax take a look and let me know if it needs to be tweaked.

charlespax, 2015-03-23:
Check out https://useiconic.com/open#icons, a bunch of free icons on github https://github.com/iconic/open-iconic.

paxmanchris, 2015-03-23:
Interesting but, I don't see any that fit what you want. Here is a mock up of the front panel with logos:

[image]

charlespax, 2015-03-23:
Icons from Google https://github.com/google/material-design-icons/releases/tag/1.0.0

Cycle channels: [image]
Backlight: 
Power: [image]
Start/stop: 
Interval: [image] or [image]

Units:

charlespax, 2015-03-25:
There is also stuff form Font Awesome http://fortawesome.github.io/Font-Awesome/icons/

charlespax, 2015-03-25:
Here's a mockup on the SLA prototype.
[image]

paxmanchris, 2015-03-25:
Okay. Here is what I have so far:
[image]

charlespax, 2015-03-25:
Changes:

- Make the record circle the same size as the power button circle
- Make the temperature units button read "**°CFK**" or "**CFK**" in a single line. Do NOT do KFC :-)
- Make the delta t a little smaller. I don't know why, but it just looks really big

charlespax, 2015-03-25:
Here's a mockup with a smaller record button and a clock.
[image]

charlespax, 2015-03-25:
Empty thermometer
[image]

Filled thermometer.
[image]

charlespax, 2015-03-25:
Here's an updated screenshot
[image]

paxmanchris, 2015-03-25:
After discussing about it, here the interface with a half filled thermometer and graph without arrow.
[image]

charlespax, 2015-03-25:
Perfect!

paxmanchris, 2015-04-01:
I added a [pdf](https://github.com/PaxInstruments/t400-enclosure/blob/master/graphics/button_label_measurement_specifications.pdf) that lays out the button placement. All the buttons are separated into svg's and exorted as 600dpi png files,  [all in the repository](https://github.com/PaxInstruments/t400-enclosure/blob/master/graphics/button_label_measurement_specifications.pdf) . If the pdf looks good please close.

charlespax, 2015-04-02:
Use drawings with the white background. For each image, indicate the size of the image bounding box, not the distance to the edge. That will be much easier to measure.

paxmanchris, 2015-04-04:
I could not find the pdf of the white background for the buttons. Could you put the asset in the repo?

charlespax, 2015-04-04:
Here you go.

[image]

paxmanchris, 2015-04-04:
I updated the  [pdf](https://github.com/PaxInstruments/t400-enclosure/blob/master/graphics/button_label_measurement_specifications.pdf).

charlespax, 2015-04-05:
I think there is a problem. The clock and record icons are circles, but the document shows different height and widths for each.

paxmanchris, 2015-04-05:
I committed some the changes you requested. I know the backlight is still off, but if made square the center circle looks distorted. Please add updated buttons layout when you can

charlespax, 2015-04-05:
Post a screenshot.

charlespax, 2015-04-05:
I've uploaded the new keypad drawing in PDF PNG and DXF (vector) formats. Here is the PNG.
[image]

paxmanchris, 2015-04-05:
Here is the comparison:
[image]

Very minor. if you can tell me which is which then fine. Might make just to square it anyways.

charlespax, 2015-04-05:
I mean post a screenshot the same as I did with my feedback. I want to see what it looks like.

paxmanchris, 2015-04-06:
I think your right, and we should have the dimensions square.
[image]

charlespax, 2015-04-06:
Now the power button is messed up.

charlespax, 2015-05-05:
@paxmanchris what's the status of this?

paxmanchris, 2015-06-21:
This is done. we got the buttons :smile:

closed 2015-06-21 by paxmanchris