PaxInstruments/t400-enclosure #93: Front panel graphics

charlespax, 2015-03-23:
@paxmanchris is taking care of the front panel graphics.

The graphics can be found in https://github.com/PaxInstruments/t400-enclosure/tree/master/graphics. Let me know if you need any other graphics to get this done.

Here are the dimensions we are working with.
[image]

paxmanchris, 2015-03-23:
@charlespax  What do you think of these?

[image]

paxmanchris, 2015-03-23:
See also [label_worksheet.svg](https://github.com/PaxInstruments/t400-enclosure/blob/master/graphics/label_worksheet.svg)

charlespax, 2015-03-23:
I think using the square or circle around the logo makes the microscope logo too small. We also end up with a ton of empty space. I think something like this would solve those two problems.
[image]

It also has the Pax Instruments name, which is nice. I think that will help with brand recognition. The microscope logo would be bigger than using the enclosed version.

MLXXXp, 2015-03-23:
Is it just going to say **T400**? With all that space, why not include the function of the device, such as **T400 Thermocouple Datalogger.**

When I pick up a new device for the first time, I find it nice to be able to find some indication of its purpose.

charlespax, 2015-03-23:
I think we should have "Pax Instruments on it" to help develop the brand. I do agree though, it would be nice to say what the device does.

charlespax, 2015-03-23:
Maybe if we just use "Pax" we can add "temperature datalogger"
[image]

charlespax, 2015-03-23:
Here is the type of label we'll have.
[image]

The label on this Rigol scope has the same 7.8 mm height as the T400 label. The calipers mark off the length we have to work with on the T400.
[image]

paxmanchris, 2015-03-23:
Okay, thanks for the input. Based on what you wrote, I added "temperature datalogger" to the label, and made a few designs with it:
[image]

charlespax, 2015-03-23:
Nice. That's starting to look pretty tight. I like how you have things divided. I think we'll have to print some of these and see how they look.

Here's another option:
[image]

paxmanchris, 2015-03-24:
Okay, revision 3. In the examples below:
1. blue, white, on black with white on gray for model number
2.  blue, black on white with  white on gray for model number
3.  blue, black on white

4, 5, 6 are the same only that "Pax" is clear.

[image]

paxmanchris, 2015-03-24:
@starno @MLXXXp  any thoughts on the revision?

charlespax, 2015-03-24:
What about something like this? Can you mock this up quickly?

[image]

paxmanchris, 2015-03-24:
Okay, here:

[image]

MLXXXp, 2015-03-24:
How about 3 (blue, black on white) but with **Temperature Datalogger" in blue?

paxmanchris, 2015-03-24:
I think it does not stand out well for blue **Temperature Datalogger**

[image]

paxmanchris, 2015-03-24:
Quick mockup of the bevel going the other way:

[image]

charlespax, 2015-03-24:
Of the layered design I like this the best. However, the layering seems a bit cluttered.
[image]

Having just the T400 blocked off looks pretty good.
[image]

I also really like the simple style of this one.
[image]

MLXXXp, 2015-03-24:
**paxmanchris** wrote:
> I think it does not stand out well for blue **Temperature Datalogger**

Then I think you'd have to say that **Pax** doesn't stand out well either.

paxmanchris, 2015-03-24:
Some variations one the one we like:
[image]

1. original from last post 
2. larger temperature datalogger
3. cleaned up, the right edge of t400 and the inner edge of the I in the P are the same distance from the outer edge (about 2mm).
4. The P looked a little to close to top, so I centered it. But now the bottom T400 is no longer flush with the brand (it did not loot right with with T400 uncentered). Yet I still like this one.

Here is with the T400 flush with the logo if moved down:
[image]

MLXXXp, 2015-03-24:
Number 4 and the last one have too much white above and not enough below. It doesn't look vertically centred and gives the impression of wasted space, especially from a distance.

I prefer number 3. The top of the **P** being close to the top edge doesn't seem too bad to me. **Temperature Datalogger** could maybe be slightly larger, though not a big as number 2.

MLXXXp, 2015-03-24:
I don't know if the **PAX** logo is cast in stone but I'd consider making the **AX** larger. Right now, there could be a slight chance that some people might ignore the microscope and arm part of the **P** and see **IAX**. **INSTRUMENTS** could remain as it is, thus becoming slightly smaller overall.

charlespax, 2015-03-25:
"Temperature Datalogger" should be black. The color blue is reserved for the Pax Instruments logo and things that enhance our brand, like the enclosure color.
[image]

Other than the microscope being too close to the top, this one does look well balanced. I like that the bottom edges of the text are aligned.
[image]

This one does seem a bit too low. It looks strange without the alignment.
[image]

I think the last one looks best.
[image]

I did a rough edit to the image below. The microscope is a touch closer to the top than to the bottom, but not as close as before. "Temperature Datalogger" and "T400" are a touch larger, which I feel is too large.
[image]

@MLXXXp can you make some mockups of what you think would look good? The graphics assets can be found at https://github.com/PaxInstruments/t400-enclosure/tree/master/graphics

MLXXXp, 2015-03-25:
Now that I've seen **Temperature Datalogger** in a larger font, I'd have to say I prefer the original size.

As I've said, I don't see the **P** being close the the top as being a problem, so I like the vertical balance of the second one above. I took that one and moved the **T400** so it's closer to the same distance from the right edge as **PAX** is from the left. I also moved **Temperature Datalogger** slightly to make it look centered.

[image]

Have you considered switching the model and description? I'm not sure which I prefer.

[image]

charlespax, 2015-03-25:
Both of those look good. The T400 on the right visually appeals to me a bit more. However, I think the T400 in the middle reads better: "Pax Instruments T400 temperature datalogger" is how I've been talking about he device. Let's go with the second option.

@paxmanchris upload an SVG with 600 dpi PNG and JPG files. I also need a PDF indicating the dimensions of the label. Let's take the label in 0.3 mm, so we'll have 72 mm x 7.5 mm.

charlespax, 2015-03-25:
I've ordered labels. Closing issue. We can make a new issue if there is a problem.

closed 2015-03-25 by charlespax