PaxInstruments/t400-enclosure #95: Top panel recess

charlespax, 2015-03-25:
Cross posted from https://github.com/PaxInstruments/t400-electronics/issues/17

There may be a problem with how deep the connector inserts into the T400.

Here is the current connector version.
[image]

The thermocouple would have to go a minimum of 5.7 mm into the connector to make contact. The shortest thermocouple connector I've seen is about 10 mm. Those numbers work well if the male part is fully inserted into the female part.
[image]
[image]

For the T400 a male thermocouple connector must pass through the enclosure and a gap before it inserts into the female connector. In this image the black line marks off the point of contact.
[image]

We could have an indentation in the top panel, so the male connectors would only have to pass through the thickness of the enclosure. That would get the job done, but isn't the best contact and may not be very secure.
[image]

We could remove the tab from the entrance of the connector and have two tabs on the sides. This would allow us to hang the connector over the edge and span the gap or part of the gap.
[image]
[image]

This would, however, cause some assembly problems when installing the top panel. I think we could overhang about 1.5 mm without causing problems. Perhaps a combination of wings on the sides and an indentation in the top panel is the solution.

Update: I accidentally removed the flange from the entrance. We would probably have a flange going up and a flange going down to aid thermocouple insertion.

starno, 2015-03-25:
Are you proposing to make a recess on the top panel to give the connector pins more guaranteed depth inside the mating connector?

charlespax, 2015-03-25:
Yes

charlespax, 2015-03-25:
It looks like a recess would be too close to the edge. We would have to make the whole top panel one flat piece. I hope we can address by changing the connector. We could also extend the PCB, but that wouldn't be as clean.

[image]

starno, 2015-05-04:
Here is the current arrangement with the new SMT connectors:

Top panel no longer has channels for the male end of the thermocouple connector, and sits 0.48mm from the inner surface of the top panel:
[image]

The shortest connector I have is the Chinese knockoff which has 10mm long male contacts.  Approximately 2.5mm of that will be going through the top panel and clearance before the connector, with the remainder inside the connector.  The bottom edge of the inlet also aligns with the bottom edge of the connector, so all the force after insertion will be displacing the spring portion of the connector:
[image]

I think this issue can be closed.

charlespax, 2015-05-04:
I don't see those top panel changes in git hub in either starno or master.

starno, 2015-05-04:
Hmm, it was part of this commit:
https://github.com/PaxInstruments/t400-enclosure/commit/2cebe68f543333da7238f13286a397d0a8f6d0ea

charlespax, 2015-05-04:
Oh I see. I have to `git checkout starno` and `git pull` to get the updates. If I'm on `master`, `git pull` does not bring in updates in `starno` and does not tell me that it was updated. I have the changes now. Closing issue.
[image]

closed 2015-05-04 by charlespax