PaxInstruments/t400-enclosure #97: Button height

charlespax, 2015-03-25:
As we discussed, the button height could be reduced. However, the current height has been growing on me. I'm still not sure what we should do.

[image]

What do you think of reducing the height on only the power button? This could avoid the device accidentally turning on while stored. Thoughts?

starno, 2015-03-29:
At the moment I think they are a little too big and a little too easy to accidentally press while holding it.  Slightly recessed power button is interesting idea.  We can probably make it subtle enough that it has the preventative functionality without the user recognizing a difference in height

charlespax, 2015-04-02:
You have the other buttons of various height. Which do you like best?

starno, 2015-04-05:
I think 1.0mm above the enclosure surface feels the best.

charlespax, 2015-04-07:
I like the feel of 1.0 mm also. Let's go with that.

starno, 2015-08-08:
1.0mm it is

[image]

commit f2d8ade1 (2015-08-08): Adjusted button height above front panel surface to be 1mm (fixes #97)
closed 2015-08-08 by starno