PaxInstruments/t400-enclosure #107: Thermocouple slots

charlespax, 2015-05-04:
I measure the thermocouple slots at about 1.5 mm:
[image]

This is probably a little too big. With the new PCB thermocouple connector a male thermocouple can flatly hit the tab while being inserted.
[image]

I believe the PCB connector is designed to accept thermocouple up to 1.1 mm, is a little larger than any thermocouple connector I've seen. I think we can close the slot a bit to prevent the male TC connector from hitting the tab and also protecting the PCB connector from being damaged by overly large male TC contacts.

starno, 2015-08-08:
@charlespax is this issue still relevant with the latest connectors?

charlespax, 2015-08-08:
I don't think it would be a problem, but I would like to tighten up that dimension to 1.3 mm with the bottom edge of the hole aligning with the inside bottom edge of the connector.
[image]

starno, 2015-08-08:
done. 

[image]

commit cdf69770 (2015-08-08): Tightened up thermocouple connector hole size to avoid possible interference when inserting (fixes #107)
closed 2015-08-08 by starno