PaxInstruments/t400-enclosure #108: Front label recess

charlespax, 2015-06-18:
The recess for the front panel sticker should be deeper. I measure 0.35 mm for the sticker thickness. Can we do a touch deeper than that?

starno, 2015-08-08:
How deep do you want to go, 0.5mm?  I'm already concerned this is going to be a problematic area because its already bending the design rules about keeping a nominal thickness and the bezel needs to be a regular shape or it will just be an eyesore.

[image]

charlespax, 2015-08-08:
How deep is it now? How do you feel about 0.35 mm or 0.4 mm, so the label is at least flush?

starno, 2015-08-08:
It was at 0.25mm, I made it 0.4mm because we can always make the depth shallower if problems arrise.

[image]

commit ddc11c08 (2015-08-08): adjusted logo sticker depth to 0.4mm so the sticker stays recessed at all times (fixes #108)
closed 2015-08-08 by starno