PaxInstruments/t400-enclosure #120: LCD bezel

charlespax, 2015-08-14:
We should tighten up the bezel. The LCD is recessed a little, so a user is able to see the edges of the LCD.
[image]
[image]
[image]

starno, 2015-08-15:
bring in 1mm on each side

starno, 2015-08-18:
brought in the bezel 1mm.  LCD border outline for reference:

[image]

commit 4a9d95c8 (2015-08-18): Made bezel on front larger, made logo depression larger (fixes #120, #118)
closed 2015-08-18 by starno