PaxInstruments/t400-enclosure #121: Tilting bail snap open

charlespax, 2015-08-14:
We should rethink the tilting bail a bit. The tilting bail has a really nice snap when it's closed. On the CNC parts the snapping open is a bit squishy. The wings don't snap into the track very well. However, the features on the bail and battery cover that are supposed to prevent over extension actually work quite nicely as a snap open feature.

Here is where we intended the bail to stop; where the nubs enter the track. You can see the highlighted features. The feature on the battery cover resists the bail from going further.
[image]

With a small amount of force I can snap the bail feature over the battery cover feature. This gives a stable hold.
[image]

This causes the battery cover to flex a bit.
[image]

starno, 2015-08-15:
mockup wingless version and the current design with better snaps, reliefs, and reinforcements

charlespax, 2015-08-27:
We'll get a CNC set of the alternate parts.

closed 2015-08-27 by charlespax