PaxInstruments/t400-enclosure #122: Threaded insert supports

charlespax, 2015-08-14:
The threaded insert ont he battery cover is not held on by much. We should add some support ribs.
[image]

starno, 2015-08-18:
added 3 ribs where there was room.  its possible to get a 4th rib if we move the thermocouple panel keyway to either side a few cm.  @charlespax what are your thoughts?

[image]

charlespax, 2015-08-18:
Moving the keyway is okay as long at is doesn't line up with a keyway on the front.

starno, 2015-08-18:
4th rib added and keyway shifted over 12mm.
[image]

The slot never actually touches the battery cover every, so it can be anywhere on that side
[image]

commit 1ea54625 (2015-08-18): added 4th rib to threaded insert boss, shifted thermocouple panel keyway over 12mm from center (fixes #122)
closed 2015-08-18 by starno