PaxInstruments/t400-enclosure #123: MicroUSB hole

charlespax, 2015-08-14:
I'd like to tighten up the MicroUSB hold. I think that would improve the look.
[image]

starno, 2015-08-18:
I dont think we can get any tighter than we are at already because there are the tabs that flare out in all four directions.  @charlespax what exactly do you want to do here?

charlespax, 2015-08-18:
What can you do to contour the hole around the connector shape? On the front piece it should be pretty easy to round off the corners. On the back piece we would have some sharp corners on the piece that sticks up.

We could change the shape of the hole a little, so there are no sharp corners. In the image below the yellow line shows the contour of the front and green shows the back. What do you think about this solution?

[image]

starno, 2015-08-18:
Looks good.  implementing...

starno, 2015-08-19:
USB slot looks a lot less blocky now.  Same amount of clearance around connector except the overall shape of the hole resembles the connector better. 
[image]

 I also changed the SD slot to be similar.  @charlespax what are your thoughts on the SD slot, better now or the way it was?

[image]

charlespax, 2015-08-19:
Looks really good in the first screenshot. The second one didn't upload, but I can see from the first one the SD slot looks good too.

starno, 2015-08-19:
oops

[image]

closed 2015-08-19 by starno

charlespax, 2015-08-19:
Yeah, looks good :-)