PaxInstruments/t400-enclosure #127: Tiny sliver edge

charlespax, 2015-08-26:
From commit 73edde1ac90ea6bb947693806ebfc6a88ce5d0d4, would this sliver cause any problems for the tooling?

This is the battery cover.
[image]

Maybe this tiny feature on the tip can't be machined the way it is modeled.
[image]

charlespax, 2015-08-26:
I think issues like this would be brought up when the factory evaluates the design. This should not hold up the 0.6 SLA print. Moving to production milestone.

charlespax, 2015-08-26:
There are other similarly thin features on the battery cover.

[image]

starno, 2015-08-27:
at the moment there are a few areas with small features like this.  They are needed to satisfy the draft angles and pull directions.  We can work with the tooling house to optimize where the parting line is and how to minimize the appearance

charlespax, 2015-10-15:
No issue form factory.

closed 2015-10-15 by charlespax